timelab/ADEM #3: Use the Scheduler library

lebeno, 2015-12-04:
We could use the Scheduler library to create separate "threads":
* sendBuffer()
* GPS data reading and processing
* analog sampling of dust sensor

lebeno, 2016-03-02:
kavers1 is working on a scheduling library

kavers1, 2016-03-07:
First version is available. Although it may crash when turning logging on, it functioned well for over 12 hours when just doing its job. Need further testing but is ready for initial use and keep the project going.

dagwieers, 2016-03-20:
@kavers1 We are having issues using SoftwareSerial (GPS) together with interrupt handlers for the PPD42.I assume that when we are receiving GPS data interrupts for PPD42 are disabled ? We always get 0 (both for PM10 as PM25). It only works if we stop using SoftwareSerial.

Does the scheduler somehow help with this ? Or do we plan to use the real UART for the communication with the GPS ?
Can we maybe use an i2c-based GPS module instead (e.g. http://www.flytron.com/sensors/180-i2c-gps-for-multiwii-and-others.html) ?

dagwieers, 2016-04-14:
We can close this issue. The TickerSchedlr is integrated in the main code now.
Next step is to implement the pseudo-code using TickerSchedlr.

closed 2016-04-14 by dagwieers