timelab/ADEM #6: create githug organization and team

lebeno, 2016-03-03:
The ADEM repository is currently under my personal github account. It would be nice to create a github "timelab" organization and an "ADEM" team in github. Unfortunately @timelab is already taken.

lebeno, 2016-03-03:
It possible, could user @timelab contact me to clarify if he/she is in any way connected to http://timelab.org, the Fablab active in Ghent, Belgium?

dagwieers, 2016-03-09:
I opened a question to Github to see what is possible. Either have them contact the user to clarify the situation, or free the account so it can be used by Timelab. I doubt the notifications are going anywhere...

dagwieers, 2016-03-09:
@lebeno Can you please unfollow him ? So it correctly appears the account is stale and unused.

lebeno, 2016-03-09:
done -> 0 followers

lebeno, 2016-03-09:
Thanks to Arnaud, Evi and Dag the timelab github account/organisation is a fact! I have transferred the ADEM repo from my personal account to the timelab organisation.

closed 2016-03-09 by lebeno