timelab/ADEM #11: i2c-gps does not build

dagwieers, 2016-08-21:
I get the following error when building i2c-gps:
```
i2c-gps:934: error: expected constructor, destructor, or type conversion before ';' token
 blink_update();   
               ^
i2c-gps:938: error: expected unqualified-id before 'if'
 if (_watchdog_timer != 0)
 ^
exit status 1
expected constructor, destructor, or type conversion before ';' token
```

dagwieers, 2016-08-21:
I think it is related to the following statements:

    #pragma region while serial available

    #pragma endregion

kavers1, 2016-08-21:
Dag,

bij mij compileerd dit zonder enige wijziging. Welke versie van arduino 
tool chain gebruiken we ? Ik ben op 1.6.7 (niet de laatste weet ik)

/De schets gebruikt 6.986 bytes (22%)  programma-opslagruimte. Maximum 
is 30.720 bytes.//
//Globale variabelen gebruiken 478 bytes (23%) van het dynamisch 
geheugen. Resteren 1.570 bytes voor lokale variabelen. Maximum is 2.048 
bytes.

/Heb je bij het compileren het board veranderd ? en eventueel het 
library path verzet naar de arduino libraries ? (moet ik doen op windows 
weet niet of dat bij linux nodig is)

koen

dagwieers, 2016-09-28:
Problem is related to a missing/incomplete config.h.

PR #24 (2016-09-30): Use platformIO with ESP8266 in Travis
This implements:
- PlatformIO support for the ADEM project
- PlatformIO support for the I2C-GPS sub-project
- Fixes the I2C-GPS build due to missing config.h
- Fixes the I2C-GPS build due to incorrect order of function definitions

This fixes #11
closed 2016-09-28 by dagwieers